dougsm/ggcnn #5: something wrong with the 'evaluate.py'

jinhuan-hit, 2019-03-19:
Hi,I have recentlt read your interesing work "Closing the Loop for Robotic Grasping: A Real-time, Generative Grasp Synthesis Approach" and have been trying to validate your results for a long time.However,I meet an error as below:
190317_1538__ggcnn_9_5_3__32_16_8       1       368.0/1750.0    21.03%
No pre-computed values.  Computing now.
190317_1538__ggcnn_9_5_3__32_16_8       2       548.0/1750.0    31.31%
No pre-computed values.  Computing now.
190317_1538__ggcnn_9_5_3__32_16_8       3       763.0/1750.0    43.60%
No pre-computed values.  Computing now.
190317_1538__ggcnn_9_5_3__32_16_8       4       941.0/1750.0    53.77%
No pre-computed values.  Computing now.
190317_1538__ggcnn_9_5_3__32_16_8       5       1071.0/1750.0   61.20%
No pre-computed values.  Computing now.
190317_1538__ggcnn_9_5_3__32_16_8       6       1047.0/1750.0   59.83%
No pre-computed values.  Computing now.
190317_1538__ggcnn_9_5_3__32_16_8       7       1165.0/1750.0   66.57%
No pre-computed values.  Computing now.
190317_1538__ggcnn_9_5_3__32_16_8       8       1145.0/1750.0   65.43%
No pre-computed values.  Computing now.
190317_1538__ggcnn_9_5_3__32_16_8       9       1157.0/1750.0   66.11%
No pre-computed values.  Computing now.
Traceback (most recent call last):
  File "evaluate.py", line 188, in <module>
    run()
  File "evaluate.py", line 156, in run
    model = load_model(model_checkpoint_fn)
  File "/home/jinhuan/anaconda3/envs/tf-gpu-27/lib/python2.7/site-packages/keras/engine/saving.py", line 417, in load_model
    f = h5dict(filepath, 'r')
  File "/home/jinhuan/anaconda3/envs/tf-gpu-27/lib/python2.7/site-packages/keras/utils/io_utils.py", line 186, in __init__
    self.data = h5py.File(path, mode=mode)
  File "/home/jinhuan/anaconda3/envs/tf-gpu-27/lib/python2.7/site-packages/h5py/_hl/files.py", line 394, in __init__
    swmr=swmr)
  File "/home/jinhuan/anaconda3/envs/tf-gpu-27/lib/python2.7/site-packages/h5py/_hl/files.py", line 170, in make_fid
    fid = h5f.open(name, flags, fapl=fapl)
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py/h5f.pyx", line 85, in h5py.h5f.open
IOError: Unable to open file (unable to open file: name = 'data/networks/190317_1538__ggcnn_9_5_3__32_16_8/epoch_00_model.hdf5', errno = 2, error message = 'No such file or directory', flags = 0, o_flags = 0)

When I run the file evaluate.py,it is ok at the begining and could print something.However,when it runs at epoch 10,it does not working well and print some errors as shown above.After then,I add a new file named epoch_00_model.hdf5 which is modified from the others into /networks/*.It continues to work.I do not know why.Could you please tell me some details about it?I am looing forward to your reply.

dougsm, 2019-03-19:
Hi there! 
Sorry about that... it looks like I was lazy with the file parsing and it only works up to epoch 99 (and is being confused by epoch 100).  I'll fix the code properly, but for the time being if you delete epoch 100 it should be fine.   There's unlikely to be any change after about epoch 50 anyway.

jinhuan-hit, 2019-03-20:
Thank you for your reply!

closed 2019-03-20 by jinhuan-hit