dougsm/ggcnn #8: AttributeError: 'NoneType' object has no attribute 'TF_NewStatus'

weiwuhuhu, 2019-05-02:
Hi,I have recently read your igreat work "Closing the Loop for Robotic Grasping: A Real-time, Generative Grasp Synthesis Approach" and have been trying to validate your results for a long time.However,I meet an error as below:
2019-05-02 11:10:37.457694: I tensorflow/core/common_runtime/gpu/gpu_device.cc:977] Creating TensorFlow device (/gpu:0) -> (device: 0, name: GeForce GTX 1050 Ti, pci bus id: 0000:01:00.0)
Traceback (most recent call last):
  File "evaluate.py", line 196, in <module>
    run()
  File "evaluate.py", line 173, in run
    succeeded, failed = calculate_iou_matches(grasp_positions_out, grasp_angles_out, bbs_all, no_grasps=NO_GRASPS, grasp_width_out=grasp_width_out)
  File "evaluate.py", line 102, in calculate_iou_matches
    gt_bbs = BoundingBoxes.load_from_array(ground_truth_bbs[i, ].squeeze())
IndexError: index 0 is out of bounds for axis 0 with size 0
Exception ignored in: <bound method BaseSession.__del__ of <tensorflow.python.client.session.Session object at 0x7f479f396a58>>
Traceback (most recent call last):
  File "/home/robot/anaconda3/envs/weiwuhuhu/lib/python3.5/site-packages/tensorflow/python/client/session.py", line 587, in __del__
AttributeError: 'NoneType' object has no attribute 'TF_NewStatus'

Firstly, I run the file evaluate.py,it occurs that:
  File "/home/robot/anaconda3/envs/weiwuhuhu/lib/python3.5/site-packages/h5py/_hl/files.py", line 170, in make_fid
    fid = h5f.open(name, flags, fapl=fapl)
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py/h5f.pyx", line 85, in h5py.h5f.open
OSError: Unable to open file (unable to open file: name = '/home/douglas/dev/ae_grasp_prediction/data/datasets/dataset_rotated_width_zoom_171219_1516.hdf5', errno = 2, error message = 'No such file or directory', flags = 0, o_flags = 0)

Sencondly,i replace the  'dataset_rotated_width_zoom_171219_1516.hdf5' with the 'dataset_190502_1033.hdf5'.
Finally, i run the evalute.py the, the obove wrong:
AttributeError: 'NoneType' object has no attribute 'TF_NewStatus' has occured . 

How can i resolve it. Thank you very much

dougsm, 2019-05-02:
Hi,
So it looks like there's two different issues. The first is the missing hdf5 file (which I think you've resolved by creating your own and changing the path, is that correct?).
The second one looks like it's potentially an issue with your tensorflow/keras install, which I'm not so sure about, but seems to come up in a quick google, e.g. https://github.com/tensorflow/tensorflow/issues/8652

weiwuhuhu, 2019-05-02:
I didn't train with the train.py
Firstly, I run the download_pretrained_ggcnn.sh to generate the dataset_190502_2135.hdf5 in dataset file,

Secondly ,i run the evluate.py, then occur the error:
IOError: Unable to open file (unable to open file: name = '/home/douglas/dev/ae_grasp_prediction/data/datasets/dataset_rotated_width_zoom_171219_1516.hdf5', errno = 2, error message = 'No such file or directory', flags = 0, o_flags = 0)
So i modify the path and i dont't find the dataset_rotated_width_zoom_171219_1516.hdf5, so i replace it with the dataset_190502_2135.hdf5 in the _dataset.txt.

Finally, i run the evaluate.py, the error occur is that:
Traceback (most recent call last):
  File "evaluate.py", line 196, in <module>
    run()
  File "evaluate.py", line 173, in run
    succeeded, failed = calculate_iou_matches(grasp_positions_out, grasp_angles_out, bbs_all, no_grasps=NO_GRASPS, grasp_width_out=grasp_width_out)
  File "evaluate.py", line 102, in calculate_iou_matches
    gt_bbs = BoundingBoxes.load_from_array(ground_truth_bbs[i, ].squeeze())
IndexError: index 0 is out of bounds for axis 0 with size 0

This is my step, i didn't run the train.py, can i run the evaluate.py directly? i can't get the result...
And my environment is :
tensorflow1.10
keras:2.0.6
Thank you very much!

weiwuhuhu, 2019-05-03:
Hi, I try run the train_ggcnn.py,the error is :
ValueError: Input 0 is incompatible with layer conv2d_1: expected ndim=4, found ndim=2
Could you please tell me some details about it?I am looing forward to your reply.

weiwuhuhu, 2019-05-03:
> Hi,
> So it looks like there's two different issues. The first is the missing hdf5 file (which I think you've resolved by creating your own and changing the path, is that correct?).
> The second one looks like it's potentially an issue with your tensorflow/keras install, which I'm not so sure about, but seems to come up in a quick google, e.g. [tensorflow/tensorflow#8652](https://github.com/tensorflow/tensorflow/issues/8652)

Thank you very much for your help.And I solve this problem recently. You are right, and the path of my cornell dataset is wrong.And I run the evaluate.py , then i get the reult such as :
190503_1859__ggcnn_9_5_3__32_16_8	5	1051.0/1760.0	59.72%

weiwuhuhu, 2019-05-03:
> Hi,
> So it looks like there's two different issues. The first is the missing hdf5 file (which I think you've resolved by creating your own and changing the path, is that correct?).
> The second one [...]

Thank you very much for your help.And I solve this problem recently. You are right, and the path of my cornell dataset is wrong.And I run the evaluate.py , then i get the reult such as :
190503_1859__ggcnn_9_5_3__32_16_8	5	1051.0/1760.0	59.72%
How can i input the image to get the result of your paper (bounding box)

TangBoheng, 2019-05-07:
I download the data from [http://s.dougsm.com/ggcnn_rss/](url) just like' download_pretrained_ggcnn.sh' do.After I unzip it I can't find  'dataset_rotated_width_zoom_171219_1516.hdf5' or 'dataset_190502_1033.hdf5'. Is there something I done wrong?

dougsm, 2019-05-08:
Hello!
If you're still having trouble, have a look at the new version that I just pushed.  It will require a little bit of set-up (it uses PyTorch rather than Tensorflow), but should be a lot more user friendly.  There might still be some issues to work out, so let me know if there is problems.

closed 2019-05-08 by dougsm