dougsm/ggcnn #12: deal with the data

wjydou, 2019-05-29:
when I download the and extract Cornell Grasping Dataset into a single directorya,and I convert the PCD files to depth images by running 
python -m utils.dataset_processing.generate_cornell_depth  data/cornell
Nect,by running train_ggcnn.py ,and have the error
INFO:root:Loading Cornell Dataset...
Traceback (most recent call last):
  File /ggcnn/train_ggcnn.py", line 269, in <module>
    run()
  File "/ggcnn/train_ggcnn.py", line 202, in run
    include_depth=args.use_depth, include_rgb=args.use_rgb)
  File "/ggcnn/utils/data/cornell_data.py", line 26, in __init__
    raise FileNotFoundError('No dataset files found. Check path: {}'.format(file_path))
FileNotFoundError: No dataset files found. Check path: /data/cornell

wjydou, 2019-05-30:
And run the train_ggcnn.py of dataset jacquard.Then have the error
Traceback (most recent call last):
  File "/ggcnn/train_ggcnn.py", line 269, in <module>
    run()
  File "/ggcnn/train_ggcnn.py", line 198, in run
    Dataset = get_dataset(args.dataset)
  File "/ggcnn/utils/data/__init__.py", line 9, in get_dataset
    raise NotImplementedError('Dataset Type {} is Not implemented'.format(dataset_name))
NotImplementedError: Dataset Type jaquard is Not implemented

wjydou, 2019-05-30:
And run the train_ggcnn.py of dataset jacquard.Then have the error
Traceback (most recent call last):
  File "/ggcnn/train_ggcnn.py", line 269, in <module>
    run()
  File "/ggcnn/train_ggcnn.py", line 202, in run
    include_depth=args.use_depth, include_rgb=args.use_rgb)
  File "/ggcnn/utils/data/jacquard_data.py", line 27, in __init__
    raise FileNotFoundError('No dataset files found. Check path: {}'.format(file_path))
FileNotFoundError: No dataset files found. Check path: /data/jacquard

dougsm, 2019-06-10:
Hi,
Can you post the whole command that you are running? It looks like you are not passing in or passing in incorrectly the dataset type and path arguments.

wjydou, 2019-06-11:
@dougsm l run in the pycharm,and gave the path of dataset .Then,the result is as shown above

dougsm, 2019-06-11:
Is the path definitely correct? Is it definitely `/data/jacquard` and not `data/jacquard`?

WuXingrui, 2019-06-17:
@wjydou You can try to delete '*' in the 22nd line of  jacquard_data.py which is in utils/data/

wjydou, 2019-06-18:
@WuXingrui Thanks for you,I had delete  '*' in the 22nd line of jacquard_data.py which is in utils/data/,but it  is also useless.

dougsm, 2019-07-10:
Hello, did you resolve this issue?
I did notice that the instructions may have been ambiguous.  The expected path format for the cornell dataset is that the 01, 02, 03, etc folders are still in the structure.  I have updated the readme.

wjydou, 2019-07-11:
@dougsm Thanks!

closed 2019-07-11 by wjydou

seivazi, 2020-03-23:
> Hello, did you resolve this issue?
> I did notice that the instructions may have been ambiguous. The expected path format for the cornell dataset is that the 01, 02, 03, etc folders are still in the structure. I have updated the readme.

@dougsm I don't understand above very well. Could you explain it one more time?
Let's say I have cornell_dataset folder then as sub-folders I should have 01,02,03,... am I right?  For train then I should provide the path to ./cornell_dataset/
Do you load each sub-folder during train? or should I have only one folder e.g.01 with all images inside

dougsm, 2020-03-23:
Yes, that's right, the directory should contain the 01, 02, etc subdirectories. Judging by the line that loads the files, every subfolder is loaded:
https://github.com/dougsm/ggcnn/blob/ad48bc5f768fe0a9ba9fd47729638e0aed46e47b/utils/data/cornell_data.py#L22