dougsm/ggcnn #14: dim error when run train_ggcnn.py

frinklai, 2019-07-08:
Hi, I forked the repository from [branch RSS2018(https://github.com/dougsm/ggcnn/tree/RSS2018).
 I already completed the step 1, 2 and 3  in [Training](https://github.com/dougsm/ggcnn/tree/RSS2018#training), which is described in readme.md

But it was an error when I run python3 train_ggcnn.py
**Input 0 is incompatible with layer conv2d_1: expected ndim=4, found ndim=2**
Can you tell me how to solve the problem?

dougsm, 2019-07-08:
Hello,
I just ran through all of the scripts and was unable to reproduce this.  The issue is that the depth image that you are passing into the network for training is not the correct size.  
First up, to make sure the dataset was created correctly, can you please print out `x_train.shape` in train_ggcnn.py and make sure that it is `(N, 300, 300, 1)`, where N is the size of the dataset.

frinklai, 2019-07-10:
Hi, thanks for your quick reply
I just regenerated dataset by run `python3 generate_dataset.py ` and I found it was an error below

**Traceback (most recent call last):
  File "generate_dataset.py", line 151, in <module>
    f.create_dataset('%s/%s' % (tt_name, ds_name), data=np.array(dataset[tt_name][ds_name]))
MemoryError**

It seems that the problem is RAM out of memory.
Are you have any idea to solve this problem ?

by the way, my pc's memory are below:
`iarc@titan:~/yc/ggcnn/ggcnn$ free -g`
              **total        used        free      shared  buff/cache   available
Mem:             15           1          13           0           0          13
Swap:            15           0          15**

dougsm, 2019-07-10:
Hi,
If you are having issues with generating the dataset, I would strongly recommend using the master branch which contains a much more user-friendly and more up-to-date implementation.
The network and training data is the same, but one of the major improvements was removing the need to pre-generate a large dataset. Instead, the data is read from disk as required.

frinklai, 2019-07-13:
ok, I got it, thank you very much for your help.

closed 2019-07-15 by dougsm