dougsm/ggcnn #16: Problem converting tiff files

OliviaNocentini, 2019-07-22:
Hi,
First thanks for the code of the GGCNN. 
I was trying to convert the dataset to achieve depth images. The problem is that when I run the command:
python -m utils.dataset_processing.generate_cornell_depth <Path To Dataset>
changing the path to my own dataset path, I get no error in the conversion, but then the d.tiff files are empty.
Could you please help me solving this problem? Thank you

dougsm, 2019-07-22:
Hi Olivia,

Just to check, try running the below script with one of the images.  If I open the .tiff files in ubuntu, they look empty, but in reality the data is there.  I think maybe it doesn't understand what to do with the floating point format.

Let me know if this doesn't work.

```
import matplotlib.pyplot as plt
import imageio

i = imageio.imread('pcd0100d.tiff')
plt.imshow(i)
plt.show()
```

Left is how it looks in the ubuntu image viewer, right is the plot from the above script.

[image]

OliviaNocentini, 2019-07-23:
Hi, thank you for answering so quickly!it works, so maybe, as you said, there was a problem with the floating point format.

closed 2019-07-23 by dougsm