dougsm/ggcnn #23: evaluation.calculate_iou_match() is something wrong？

Cheniitt, 2019-12-10:
when training the ggcnn2, validating is always zero, but the loss is  Decreasing.

Cheniitt, 2019-12-10:
[image]

dougsm, 2019-12-13:
g'day @Cheniitt 

It definitely looks like there's something strange going on.  Can you provide more details?  What dataset are you using? What was the command you ran? If you run it with the `--vis` flag does it look like the network is learning properly?

Cheniitt, 2019-12-13:
thanks for your reply @dougsm 
I used the jacquard dataset to train ggcnn2, run the command `python train_ggcnn.py --description training_example2 --network ggcnn2 --dataset jacquard --dataset-path <Path To Dataset> --vis`, when training with the flag `--vis`, it looks the network is learning properly.

Cheniitt, 2019-12-13:
I found in https://github.com/dougsm/ggcnn/blob/ad48bc5f768fe0a9ba9fd47729638e0aed46e47b/utils/dataset_processing/grasp.py#L269,Whenever it always returns 0.

closed 2019-12-18 by Cheniitt