dougsm/ggcnn #24: About the code

ueuuop8591, 2020-02-16:
q_img = q_img.cpu().numpy().squeeze()
    ang_img = (torch.atan2(sin_img, cos_img) / 2.0).cpu().numpy().squeeze()
    width_img = width_img.cpu().numpy().squeeze() * 150.0

    q_img = gaussian(q_img, 2.0, preserve_range=True)
    ang_img = gaussian(ang_img, 2.0, preserve_range=True)
    width_img = gaussian(width_img, 1.0, preserve_range=True)

question1: What does the q_img mean? 
question2: What is the effect about "cpu().numpy().squeeze()"?

dougsm, 2020-02-16:
The network has 3 outputs, the quality (Q), the angle and the grasp width (see https://arxiv.org/pdf/1804.05172.pdf).  `q_img` corresponds to the quality output.

`cpu().numpy().squeeze()` are operations on the pytorch tensor to move the data from the gpu to cpu turn it into a 300x300 numpy array before doing any further processing.  You can find out more in the pytorch documentation, e.g. cpu().numpy().squeeze()

Hope that helps :-)

ueuuop8591, 2020-02-17:
Thanks to your explain.

ueuuop8591, 2020-02-17:
@dougsm 
Dataset = get_dataset(args.dataset)
    test_dataset = Dataset(args.dataset_path, start=args.split, end=1.0, ds_rotate=args.ds_rotate,
                           random_rotate=args.augment, random_zoom=args.augment,
                           include_depth=args.use_depth, include_rgb=args.use_rgb)
    test_data = torch.utils.data.DataLoader(
        test_dataset,
        batch_size=1,
        shuffle=False,
        num_workers=args.num_workers
    )

This code in the eval_ggcnn.py is used to input pictures which is evaluated  grasp point?

ueuuop8591, 2020-02-17:
@dougsm  how about '--split' mean?

closed 2020-02-18 by ueuuop8591