dougsm/ggcnn #26: about loss

Steve-Mq, 2020-04-16:
Hello ! dougsm:
Thank you very much for the code you provided.I have a question why you choose the MSE function as the loss function.The input is an (depth) image, the ground truth is an image, how does the MSE function  ensure that the input is moving toward the label? Could you tell me your understanding?Thank you very much!
Any help will be much appreciated.

dougsm, 2020-04-20:
Hi @Steve-Mq 
The MSE is acting pixelwise on the image-based labels at the output, so the loss is computed as the pixelwise ground truths and the output grasp map (set of images).  MSE is commonly used for reconstruction tasks.  I hope that helps.

Steve-Mq, 2020-04-21:
@dougsm  Thank you very much!

closed 2020-04-21 by Steve-Mq