dougsm/ggcnn #27: Pretrained weights for pytorch version

ancorasir, 2020-05-13:
Thanks the sharing the code with the community. Would you share the pretrained weights for pytorch as well?  I'd like to test trained model on images of novel object for example.

dougsm, 2020-06-17:
Hi @ancorasir 
Sorry for the super slow reply, I completely missed this.  
Pretrained weights for some different conditions are available here: https://github.com/dougsm/ggcnn/releases/download/v1.0/ggcnn2_weights.zip

Note that if you have a specific application in mind, retraining for that scenario might provide better results, especiall aspects like field of view and data augmentation.

closed 2020-06-17 by dougsm

trungpham2606, 2020-07-02:
@dougsm 
Seem there's something wrong that I cant load your saved file.
Can you specify the pytorch version you used for this project ?