dougsm/ggcnn #28: About the output results and 6D description

SWEETYT, 2020-06-16:
dear author, thank you for your code and literature. I have run the program you provided, and the outputs are four images:RGB, Depth,Q and Angle. But the g mentioned in the article is(p, phi, w,q) .I wonder how can I get ‘g’ from the outputs(the four images). Looking for your reply.

dougsm, 2020-06-17:
Hi @SWEETYT 

To get from the pixel coordinates to you need to use the camera intrinsic matrix.  We have an example in our ROS deployment of GG-CNN here:
https://github.com/dougsm/mvp_grasp/blob/c301ba2f4b0c0fb0c92b61a34f2821e56c693163/ggcnn/ros_nodes/ggcnn_predict_rt.py#L95

If you're unfamiliar with using the camera intrinsics, this link has a good overview of what that means:
https://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html

SWEETYT, 2020-06-24:
dear author,thank you very much for your reply.I already know how to get six dimensional pose params. 
And I have another question:If I use my own RGB image and depth image, can I also get the grasp box through the network test you provided. But I checked eval_ggcnn.py , Cornell dataset and Jacquard dataset provide not only RGB img and depth img, but also the coordinates of grasp boxes. Is it necessary to mark the graspable box and provide the coordinates when I test? 
Thank you for your patience.Looking for your reply.

dougsm, 2020-07-03:
If you want to evaluate on your own dataset, you will need to provide ground truth grasp rectangles.  The easiest way to to this would be to use the same format as the Cornell or Jacquard datasets, then you can use the existing code for evaluation.  
If you don't want to evaluate, but just see the output on some custom images, then you don't need the ground truth data. You can just get the outputs by doing a forward pass of the network, then use `dataset_processing.grasp.detect_grasps` to get the best grasps and visualise them using `dataset_processing.grasp.Grasp.plot` onto your custom images.
Hope that helps :)

SWEETYT, 2020-07-04:
> If you want to evaluate on your own dataset, you will need to provide ground truth grasp rectangles. The easiest way to to this would be to use the same format as the Cornell or Jacquard datasets, [...]

Thank you very much for your reply. I have solved this problem.

closed 2020-07-06 by dougsm

aakaash06, 2023-03-14:
> Thank you very much for your reply. I have solved this problem.

Hey, can you please provide the code for this?