dougsm/ggcnn #30: Pytorch weights

trungpham2606, 2020-07-06:
@dougsm 
There's something wrong with your saved pytorch model.
It seems that only you can load the file with torch.load().
[image]

for example in your weight file, what is the ggcnn_large ? 
can you please send me the weights file only (i mean the state_dict of weights from model) ?

dougsm, 2020-07-10:
Hi @trungpham2606 
Sorry about that. I've uploaded new weights files (both the full pytorch model and the state dicts) and added instructions to the README. Please make sure you pull the newest version of the repo just to make sure the code all matches the weights I just uploaded.

trungpham2606, 2020-07-10:
@dougsm 
Thank you so much.
Iam gonna test it right now