dougsm/ggcnn #32: Something wrong when I converted pcd files to depth images, using the python script

Zhangeaky, 2020-08-07:
Thank you for your code and paper first! 
when I used the file **generate_cornell_depth.py** with the commandline 
"python -m utils.dataset_processing.generate_cornell_depth <Path To Dataset>", the result  depth picture is strange,all result depth    
images look the same,just like this 
[image]

dougsm, 2020-08-10:
Hey @Zhangeaky 
I've seen this before. I think the problem isn't that the files are incorrect, but rather the image viewer (e.g. the default viewer in ubuntu) doesn't know how to interpret the floating point data.

To test, you can try and visualise it using matplotlib instead:
```
import matplotlib.pyplot as plt
import imageio

img = imageio.imread("<path_to_img.tiff>")
plt.imshow(img)
plt.show()
```

Hopefully that will show that the correct data has been saved.  Otherwise, shoot me a reply and I'll take another look.

Zhangeaky, 2020-08-10:
Thanks! I just solved it by your solution

closed 2020-08-10 by Zhangeaky