dougsm/ggcnn #34: About Depth Input processing in paper

kendj-staff, 2020-12-08:
Dear Doug,
when I read the paper, I have some questions about the input- depth image's process.  I have search on the internet, but I could not find an understandable answer.
I am a newbie learner on deep learning and robot, the questions I have maybe is a little silly, hope you do not mind it.
Could you help me to answer these questions? thanks a lot!!

Background: As you mentioned in your paper,  you subtract the mean of each depth image, center its value around 0 to provide depth invariance.
Question 1: I have a little confused about why we need to provide depth invariance?   
Question 2: what situation would happen if we don't do this step?

dougsm, 2020-12-08:
Hi @kendj-staff , I hope I can answer your questions:

1/2. This is mostly for the beneift of the neural network.  For training and inference, it is important that the value is close to zero, and in most tasks you would normalise data by subtracting the mean and dividing by the standard deviation.  Here we subtract the mean to zero centre the data, so that it doesn't matter the distance from the image to the object.  Otherwise, the same image from 1m away would give very different output to the same image from 1.5m away, for example.  
3. I believe that is just scaling from metres to millimetres.

kendj-staff, 2020-12-08:
oh !! I got it!!  your answer is really clear for me!!! thank you for your patiently answer explanation again!!!

closed 2020-12-08 by kendj-staff