dougsm/ggcnn #37: One problem ＞﹏＜

74284853, 2021-01-20:
Hi,teacher.I'm new to this field.I read this paper and tried to reproduce it,but meet some problems,Could you tell me how to calculation speed of single image use'eval_ggcnn'?Just like the '6ms for a single depth image' presented in the paper.I used the jacquard dataset.

HanwenCao, 2021-01-20:
Hi, I hope I can help you. But I am not 100% sure. You might want to monitor the run time of https://github.com/dougsm/ggcnn/blob/0c50aa7600e8a30d44c5c85cebd6e3394a81f30e/eval_ggcnn.py#L78
using a python timer.

74284853, 2021-01-21:
> Hi, I hope I can help you. But I am not 100% sure. You might want to monitor the run time of
> 
> https://github.com/dougsm/ggcnn/blob/0c50aa7600e8a30d44c5c85cebd6e3394a81f30e/eval_ggcnn.py#L78
> 
> 
> using a python timer.

could you tell me your email ?I look forward to further communication with you