dougsm/ggcnn #57: AttributeError

normajeane-m, 2023-09-19:
I appreciated your amazing research.

After the pcd files were converted to depth images (using function `generate_cornell_depth`), I got some errors when I tried to train the network with Cornell dataset.

```
AttributeError: Caught AttributeError in DataLoader worker process 0
AttributeError: Couldn't find function center in BoundingBoxes or BoundingBox
```

I tested the task to loading the ggcnn using pre-trained model.

> from models.ggcnn import GGCNN
> model = GGCNN()
> model.load_state_dict(torch.load('ggcnn_weights_cornell/ggcnn_epoch_23_cornell_statedict.pt'))
> 
> **All keys matched successfully**

I would like to thank you once again.

kalyanksk, 2023-10-01:
I also faced the issue and changing numpy version worked for me!
```
pip uninstall numpy
pip install numpy==1.22
```

Anmol-002, 2024-11-17:
can you please share your repository of this code @t-272  @normajeane-m  @kalyanksk