dougsm/ggcnn #59: grasp.py问题

zsgwq, 2024-09-12:
File "C:\Users\adminster\Desktop\yolo-ggcnn-main\yolo-ggcnn-main\ggcnn\train_ggcnn.py", line 253, in run
    test_results = validate(net, device, val_data, args.val_batches)
  File "C:\Users\adminster\Desktop\yolo-ggcnn-main\yolo-ggcnn-main\ggcnn\train_ggcnn.py", line 103, in validate
    val_data.dataset.get_gtbb(didx, rot, zoom_factor),
  File "C:\Users\adminster\Desktop\yolo-ggcnn-main\yolo-ggcnn-main\ggcnn\utils\data\jacquard_data.py", line 42, in get_gtbb
    gtbbs.rotate(rot, (c, c))
  File "C:\Users\adminster\Desktop\yolo-ggcnn-main\yolo-ggcnn-main\ggcnn\utils\dataset_processing\grasp.py", line 42, in <lambda>
    return lambda *args, **kwargs: list(map(lambda gr: getattr(gr, attr)(*args, **kwargs), self.grs))
  File "C:\Users\adminster\Desktop\yolo-ggcnn-main\yolo-ggcnn-main\ggcnn\utils\dataset_processing\grasp.py", line 42, in <lambda>
    return lambda *args, **kwargs: list(map(lambda gr: getattr(gr, attr)(*args, **kwargs), self.grs))
  File "C:\Users\adminster\Desktop\yolo-ggcnn-main\yolo-ggcnn-main\ggcnn\utils\dataset_processing\grasp.py", line 316, in rotate
    self.points = ((np.dot(R, (self.points - c).T)).T + c).astype(np.int64)
ValueError: shapes (2,2,1) and (2,4) not aligned: 1 (dim 2) != 2 (dim 0)

zsgwq, 2024-09-18:
这是我在训练jacquard数据集时出现的问题

t-272, 2024-10-25:
jacquard数据集可以发我一下吗

xiaoming-gong, 2024-12-17:
I'm having the same issue, have you solved it now?

xiaoming-gong, 2024-12-17:
Using python 3.8 and numpy 1.21.1 can solve this problem.